rerun-io/rerun #1937: Consider accepting backslashes as path separators in object paths

Wumpf, 2023-04-19:
Recently ran into a silly bug where a python path got converted to string and then used as an entity path

[image]

We should consider converting backslashes internally to slashes and maybe even accept `pathlib.Path`

emilk, 2023-04-21:
Backslashes are also used for escape sequences, which would make using `\` for separators confusing imho

Wumpf, 2023-04-24:
agreed, backslashes is probably a bad idea. Also, thinking about it again taking a python `Path` is misleading as well - we're not dealing with file paths, so we also shouldn't take one. We could instead provide a simple `EntityPath` class that allows operations like `parent` etc.

Wumpf, 2023-11-06:
not doing this any time soon, likely to make things worse

closed 2023-11-06 by Wumpf