rerun-io/rerun #1956: Support affine transformations (allow scaling!)

Wumpf, 2023-04-22:
Right now our transform hierarchy only has pinhole transforms and rigid transforms (== rotation+ translation).
Since our transform hierarchy internally deals with affine transformations it should be relatively easy to add those.

Caveat here is ofc that there are various ways of representing them. Straight forward and most flexible (too flexible?) is a 3x3 matrix + quaternion (this is what glam does), but users may want to specify rotation+translation+scale (careful: Need to define the order of these operations!). Making this ticket related to #1882

Wumpf, 2023-04-22:
should probably also add Mat4 support as well while on it (trickier because need to define what's supported)

emilk, 2023-04-26:
Affine transforms include rotation, translation, non-uniform scaling, and shearing.

This is best represented with a 4x3 matrix, which is effectively what glam does (https://docs.rs/glam/latest/glam/f32/struct.Affine3A.html - I think @Wumpf meant to write "3x3 + translation" not "3x3 + quaternion").

Our API should be able to specify such a 4x3 matrix directly, or as translation*rotation*scale.

Adding a fourth row (4x4 matrix) adds support for full projection matrices, but this needs careful consideration, so let's leave that for later.

emilk, 2023-05-02:
We should probably ignore the scaling for point radius when that radius is in UI points

closed 2023-05-24 by Wumpf