rerun-io/rerun #1964: Weird UI interaction with Play, Pause and Follow latest data, and Restart from beginning command

h3mosphere, 2023-04-25:
When rerun first recieves data, Follow latest data is enabled by default. This is toggled with space, which also toggles play and pause (apparently). There is some weirdness with the interaction between the four functions:

`Pausing` (by pressing space), then waiting a few frames, then `Playing` (space), resumes from the end of the timeline, not from the point it was paused at.
`Restarting` the timeline while `follow` is enabled, restarts the timeline, then immediately goes to the last value in the timeline (following the latest data)

It also is not obvious how to turn off the follow features using key-strokes. Only by clicking on the play button.

**To Reproduce**
Steps to reproduce the behavior:
1. Have some events coming in continuously
2. Press space to pause, wait. press space to resume. it resumes from end of timeline.
3. Ctrl <- to get to the start of the timeline, it immediately jumps to the end

**Expected behavior**
`Pausing` then resuming (via space) should resume from where the timeline paused.
`Restarting` should start from the beginning of timeline.
`Play` and `pause` should be toggled by space. 
`Follow` should perhaps be enabled by another keycode, or cycling with play and pause (although if feels they are just currently conflated because ... the code is currently just written that way..., and should be separated).

I'll probably have a look tomorrow and fix this, any opinions on how it should behave are appreciated.

Wumpf, 2023-04-25:
Thanks for filing this and even offering to fix it! I agree with your expected behavior:

* Play/Pause in particular definitely shouldn't jump around only because new data was added
* Restart paired with follow is obviously broken 😱
* Play/Pause toggling with space seems to work for the most part except the detour on follow..
* `Follow` behavior/shortcut a bit tricky, I'd still expect things to pause when pressing space, but the shortcut for `Follow` shouldn't be space - when I tried it again briefly it also didn't go back to follow on space which seems correct to me.

But maybe @emilk wants to add something to this? I vaguely remember there was some back and forth on this 🤔

h3mosphere, 2023-04-26:
Ok,

There is some interleaving of code dealing with states of `play` and `follow`. There's a fairly involved comment there, so this is probably to do with the back-forth, and there is some heuristics regarding if we are looking at a file, or following a stream. see the toggle_play_pause function:

https://github.com/rerun-io/rerun/blob/0ad19980be99cb2f669d38c2f1410a38206cbe74/crates/re_viewer/src/misc/time_control.rs#L304

emilk, 2023-04-26:
Previous issues:
* https://github.com/rerun-io/rerun/issues/882
* https://github.com/rerun-io/rerun/issues/941

PR #1986 (2023-04-27): Add keyboard shortcut for "Follow", and stop following on "Restart"
Stop following data on timeline restart
Bind Ctrl -> as follow.

fixes: #1964 

### What

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)

PR Build Summary: {{ pr-build-summary }}
closed 2023-04-27 by emilk