rerun-io/rerun #1967: Jump to latest added timeline whenever a new timeline is added

Wumpf, 2023-04-25:
Otherwise we're showing `log_time` all the time and this is unlikely of interest once there is a new one.

Also, this is another thing that should go to blueprint configuration!

emilk, 2023-04-26:
Some caveats:

* `log_time` has all the logged data. Any other timeline could be missing a lot of the logged data, making users wonder "why isn't my logged data showing up!?
* Switching timeline at runtime (when a new timeline is added) can be quite annoying/distruptive ("why did my UI change!?")

To make this good I think we should separate the two cases:
* Selecting a default timeline
* Automatically switching to a new timeline

For the second case, we should `re_log::info!` the switch, which will show a little toast notification to the user, making it less surprising.

We should also make sure that any Space View which lacks data _because there is none on the current timeline_ should communicate that by showing some text "No data for this view on the current timeline ${name}. Click here to switch to the `log_time` timeline instead." or similar.

Wumpf, 2024-01-09:
seems at some point we implemented this without any extra logging

closed 2024-01-09 by Wumpf