rerun-io/rerun #1988: 2D primitives Z/layering

emilk, 2023-04-27:
We don't currently define which order 2D shapes are painted. Once we support more primitives like filled rectangles, circles, text etc, it will be very important to be able to specify what is on top of what.

We also need good default, e.g. images should probably be furthest down whatever happens.
We should maybe default to painters-algorithm (whatever is logged first is painted furthest back), as that is what many users are used to from e.g. opencv.

Let's investigate nomenclature other libraries and tools use.

Wumpf, 2023-04-27:
related to
* #702 
* #1025

emilk, 2023-05-03:
Let's avoid confusion by not calling this "Z". A depth-image contains Z-values, but can also have a draw-order/layer which is completely unrelated.

emilk, 2023-05-03:
proposal: `DrawOrder(f32)` component

emilk, 2023-05-03:
We should also have a default `DrawOrder` for all the 3D stuff projected into 2D, so that you can specify:

* Paint my image
* Project 3D stuff on top of the image
* Paint these rectangles on top of everything else

PR #2138 (2023-05-17): Draw order for python api
Adds python api for draw order and improves draw order demo in api_demo for both rust and python.

Python:
[image]

Rust:
[image]

Fixes #1988
* #1988

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)

PR Build Summary: https://build.rerun.io/pr/2138
closed 2023-05-17 by Wumpf