rerun-io/rerun #1989: Segmentation Fault when running the `rerun` binary on Apple Silicon (M2)

luizberti, 2023-04-27:
**Describe the bug**
1. `cargo install rerun`
2. `rerun`
3. `SIGSEGV`

**Expected behavior**
Not segfaulting

**Backtrace**

```
2023-04-27T12:25:40Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.

    Welcome to Rerun!

    ...

    Happy Rerunning!

Rerun caught a signal: SIGSEGV

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting

   2: rerun::crash_handler::install_signal_handler::signal_handler
   3: __platform_memmove
   4: wgpu_core::device::queue::<impl wgpu_core::hub::Global<G>>::queue_write_texture
   5: <wgpu::backend::direct::Context as wgpu::context::Context>::queue_write_texture
   6: <T as wgpu::context::DynContext>::queue_write_texture
   7: wgpu::Queue::write_texture
   8: egui_wgpu::renderer::Renderer::update_texture
   9: egui_wgpu::winit::Painter::paint_and_update_textures
  10: <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::paint
  11: eframe::native::run::run_and_return::{{closure}}
  12: <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event
  13: winit::platform_impl::platform::app_state::Handler::handle_nonuser_event
  14: winit::platform_impl::platform::app_state::AppState::cleared
  15: std::panicking::try
  16: winit::platform_impl::platform::observer::control_flow_end_handler
  17: <unknown>
  ...
  26: <unknown>
  27: winit::platform_impl::platform::event_loop::EventLoop<T>::run_return
  28: eframe::native::run::run_and_return
  29: std::thread::local::LocalKey<T>::with
  30: eframe::native::run::wgpu_integration::run_wgpu
  31: eframe::run_native
Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting

fish: Job 1, 'rerun' terminated by signal SIGSEGV (Address boundary error)
```

**Desktop (please complete the following information):**
 - M2 Macbook Air (macOS 13.3.1)

emilk, 2023-04-27:
Hi! Thanks for the report. This is a crash in [`wgpu`](https://github.com/gfx-rs/wgpu) which we just updated on `main`.

Would you mind trying rerun `main`?

```
git clone git@github.com:rerun-io/rerun.git
cd rerun
cargo run -p rerun
```

luizberti, 2023-04-27:
Seems fixed 👍

closed 2023-04-27 by luizberti