rerun-io/rerun #1994: Figure out a way to safely sandbox contributor pull-requests but still get build results

jleibs, 2023-04-27:
Pull requests currently fail when submitted by external contributors because they are run without our secrets.  The build part is fine, but the upload steps end up failing.

This is generally a good thing not to leak our credentials to external contributors, but it's a bit annoying since it would be nice to end up with sample html results, etc. from CI.

I started working on a proof-of-concept to allow this via pull_request_target: https://github.com/rerun-io/rerun/pull/1991

However, there are some unresolved security implications of that approach.  We need to do more thinking and probably introduce more granular gcloud permissions to do this safely.

jleibs, 2023-04-27:
This seems like a worthwhile read: https://securitylab.github.com/research/github-actions-preventing-pwn-requests/

jleibs, 2023-04-27:
Specifically our solution should likely look like: https://docs.github.com/en/actions/using-workflows/events-that-trigger-workflows#using-data-from-the-triggering-workflow

jprochazk, 2023-09-29:
We can now run contributor PRs through our CI safely: https://github.com/rerun-io/rerun/pull/3557/checks

Remaining work:
- [ ] Clean up pull request template
  - No need for PR summary, web demo link, etc. if the contributor won't be able to use it

jleibs, 2023-11-28:
@jprochazk what has to happen to make the contributor template different?

emilk, 2024-07-03:
We've now seen one malicious (and failed) attempt at stealing our GitHub token: https://github.com/rerun-io/rerun/pull/6736.

We believe the GitHub action permission system is not granular enough to have both security, power, and easy 3rd party PRs. So we need to move off GitHub actions.