rerun-io/rerun #1995: A single call to `rerun.log` should never produce more than 1 "event" on the timeline

jleibs, 2023-04-27:
This is currently violated by the python splat handling which logs separate events for the splat vs the data.

teh-cmc, 2023-04-27:
Beyond the splat hack (#1893), I can already think of at least two legit situations (I'm sure there are many others) where the number of log calls initiated by the user and the number of rows actually stored end up differing: dumping the store to disk and back, and [bucket-level compaction](https://github.com/rerun-io/rerun/issues/512).

So I think we should handle the issue by 1) making sure that when these kinds of M:N mappings between log-calls vs. actual number of rows stored happen, all the rows involved end up sharing the same `RowId` and 2) that the UI merges elements that share the same `RowId` when displaying them.

nikolausWest, 2023-09-11:
Should this still be open?

abey79, 2025-03-05:
Should this still be open?

teh-cmc, 2025-03-05:
Nah, thinking in individual rows/events is cursed at this point.

closed 2025-03-05 by teh-cmc