rerun-io/rerun #2013: Proper Gradio Support

jleibs, 2023-05-01:
There are two complexities to figure out with Gradio:
(1) Gradio (and in particular hugging face spaces) run as a persistent service even across multiple users.  This means we need to avoid ( be **able** to avoid global session state interfering across multiple user sessions)
(2) Many gradio workloads that would be interesting feature long-running task. The real "killer app" for Rerun in Gradio is a Rerun session that gracefully handles incremental logging over the websocket interface, all running within a hugging face space. This would be vastly superior to the opaque progress bar workloads today.

TODO(@jleibs ) fill out more details

emilk, 2024-04-30:
https://x.com/radamar/status/1785066677706928326
https://huggingface.co/spaces/radames/gradio_rerun

jleibs, 2024-05-10:
Reminder to self: give the gradio component a blueprint argument that does the right thing.

jleibs, 2024-05-29:
https://github.com/radames/gradio-rerun-viewer/pull/1

jleibs, 2024-06-03:
This has been published.

closed 2024-06-03 by jleibs