rerun-io/rerun #2015: Allow (some) plots to aggregate / downsample data automatically based on zoom-level.

jleibs, 2023-05-02:
There are many cases where plots over long regions of times become too noisy to visualize nicely. In these cases it would be nice to have different policies for aggregating multiple samples into a single visualize representation.

An example from a user:
> I have a data of product sales of from a store and I would like to see bars of whole month and when zoomed in I would like to see the composition of the sold products on an hourly basis.

This is probably another area where a "System" (https://github.com/rerun-io/rerun/issues/1155) plugin could make sense. This is similar in concept to a "derived component" but changes the way we think about time and instance-identity a bit so it's worth thinking about as we scope out that work in the future.

emilk, 2023-05-02:
Downsampling is also crucial for performance when plotting 100k+ datapoints

jleibs, 2023-11-28:
Dupe of: https://github.com/rerun-io/rerun/issues/4271

closed 2023-11-28 by jleibs