rerun-io/rerun #2016: Expose timeline gap-detection to time-series plots

jleibs, 2023-05-02:
We already detect when there are large gaps in data for rendering the timeline, but our scalar plots don't have access to this data, creating an in-congruent view.

Exa:
[image]

It would be nice to expose this gap data to the plots so they can choose to add the same jumps with an appropriate visual indication. Definitely something that should be toggle-able if implemented.

emilk, 2023-05-02:
Requires us to move away from `egui::Plot`, which is something we want to do anyway for a variety of reasons

MichalLebeda, 2023-05-15:
Minimum gap setting would also be great to have exposed.