rerun-io/rerun #2032: Rerun broken on mobile Safari

emilk, 2023-05-03:
It worked very recently.

jprochazk, 2023-05-16:
No adblocker here, also broken on Android Chrome

emilk, 2023-05-16:
This is what I get on my iOS Safari in the dev console:

```
[Error] panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_render_pipeline
    Color state [0] is invalid
    Format Rgba8UnormSrgb can't be multisampled

', /home/jp/.cargo/git/checkouts/wgpu-b5b0e12eef369ba9/de497ae/wgpu/src/backend/direct.rs:3019:5
```

Three things here:
* Why isn't this shown to the user? We have code for that 🤔 
* Why is this broken? When did it break?
* `wgpu` panicking on errors is extremely annoying

emilk, 2023-05-17:
`get_parameter_i32(glow::MAX_SAMPLES)` returns `0` on my iPhone, which wgpu interprets as supporting at most 2x MSAA

emilk, 2023-05-17:
Fix:
* https://github.com/gfx-rs/wgpu/pull/3780

emilk, 2023-05-23:
Is awaiting a wgpu release

emilk, 2023-05-25:
https://github.com/gfx-rs/wgpu/pull/3780 is merged and released as part of 0.16.1, but the fix doesn't seem to take O.o

emilk, 2023-05-29:
Currently blocked on weird wasm-bindgen issue:
* https://github.com/rerun-io/rerun/pull/2206

emilk, 2023-05-30:
So it turns out wgpu 0.16.1 did _not_ include a release of `wgpu-hal`, which is still on 0.16.0. Since my fix was to `wgpu-hal`, we don't have the fix in hand yet: