rerun-io/rerun #2076: timeline handle disappears when dragged too far to the left

Wumpf, 2023-05-10:
Via kai1 on discord https://discord.com/channels/1062300748202921994/1062300748777529379/1105508608604639242

> at least in the web viewer I can drag the timeline handle off to the left (I think that's intentional to only show timeless components?) but dragging it back doesn't work.
Really weird behavior IMO given the "timeless" position is under the streams panel, and it just seems like the timeline handle has disappeared

[image]

emilk, 2023-05-15:
Can reproduce.

The problem is the code in `time_marker_ui` in `crates/re_viewer/src/ui/time_panel/mod.rs`.

In `expanded_ui` we paint the time cursor line last so it is on top of all other elements, such as the horizontal guide-lines added by the left-hand tree ui. This means it would go on top of the left-hand tree-view, except that `time_marker_ui` check for this and manually hides the time cursor if it is too far left or right. However, this means we are also not tracking the dragging of it when it is outside the "allowed" range.

If we move the call to `time_marker_ui` further up in `expanded_ui` the behavior is correct, but we get some visual artifacts from horizontal lines being painted over the vertical time cursor. Maybe this is better than the above bug though.

A more proper solution would come from implementing proper zlayering in egui. Most of the work is done (https://github.com/emilk/egui/pull/2633) but needs more testing.

teh-cmc, 2024-05-03:
This is now behaving exactly as I would expect on `main` so... closing this.

- See also https://github.com/rerun-io/rerun/issues/5418