rerun-io/rerun #2077: Play button should toggle when no data is streamed

Wumpf, 2023-05-10:
Via Kai on Discord https://discord.com/channels/1062300748202921994/1062300748777529379/1105509445137608724

In cases where we *know* no new data is streamed in (i.e. just loading a file / closed connection), the play button should toggle and not keep playing. More like a video player

emilk, 2023-05-10:
Makes sense. Related to:
* https://github.com/rerun-io/rerun/issues/1964

jleibs, 2023-05-12:
The fix for this was more complicated than anticipated and introduced a bug where the viewer would prematurely pause if playback caught up with the recording timeline during log-ingestion. Re-opening this and reverting this fix...

jleibs, 2023-05-12:
Fix for the regression: https://github.com/rerun-io/rerun/pull/2106

closed 2023-05-12 by jleibs