rerun-io/rerun #2083: Notebook "Play" mode repaints gui each frame

emilk, 2023-05-11:
Try running `examples/python/notebook/cube.ipynb` and turn on the performance metrics, and you'll see:

[image]

Something is calling `request_repaint` on egui, even though there is no new log events arriving.

emilk, 2023-05-11:
The fix should be the same as for https://github.com/rerun-io/rerun/issues/2077

emilk, 2023-05-11:
This regression happened in a recent refactor of re_viewer, where the `request_repaint` was moved

jleibs, 2023-05-12:
The fix for this regressed other behavior.
See: https://github.com/rerun-io/rerun/pull/2103

jleibs, 2023-05-12:
Fix for the regression: https://github.com/rerun-io/rerun/pull/2106

closed 2023-05-12 by jleibs