rerun-io/rerun #2108: Publish `rerun` binaries on popular repositories

teh-cmc, 2023-05-12:
See https://repology.org/ for an exact definition of "popular".

Requires #2107

emilk, 2023-05-15:
I assume you mean `apt`, `brew`, `winget` etc.

For our Python users this brings nothing to the table (`pip install rerun-sdk` already sets up a `rerun` in your `PATH`).

For our Rust SDK users this is great though, as you can get nicer `spawn`: https://github.com/rerun-io/rerun/issues/2109 without having to wait for the extreme slow compile of `cargo install rerun`.

It is also nice for people just looking for a simple image- and mesh-viewer (https://github.com/rerun-io/rerun/pull/2060)

emilk, 2023-05-19:
This would also help with:
* https://github.com/rerun-io/rerun/issues/1997
as we could then have `rerun` just depend on a few minimal features

emilk, 2023-08-11:
We should also remember to add analytics for how the user installed rerun. See:
* https://github.com/rerun-io/rerun/issues/2964

emilk, 2023-09-13:
We should consider using `cargo-dist` for this: https://github.com/axodotdev/cargo-dist