rerun-io/rerun #2142: Make a plan for how to make Blueprint ergonomic

emilk, 2023-05-16:
Starting in https://github.com/rerun-io/rerun/pull/2010 we will load the blueprint store from the data store each frame, then write it back. This has many sharp edges:

* Things that are `#[serde(skip)]` will be forgotten between frames
* We need to special-implement `PartialEq` for some thing due to `#[serde(skip)]`
* For each new element in the blueprint we need to write code to:
  * Create a new component for it OR add it to an existing `struct` and change its arrow type in the "mandatory" doc-test
  * Load it from the store
  * Diff it
  * Write it to the store
  * Write a Python interface for it
 
This will very quickly collapse under its own weight unless we come up with a better plan

emilk, 2023-05-26:
We have a plan: code-gen from Flatbuffers IDL. Generate Python and Rust SDKs including the arrow conversions.

We also have https://github.com/rerun-io/rerun/pull/2188

closed 2023-05-26 by emilk

jleibs, 2023-08-01:
Re-opening this until we have line-of-sight on the code-gen flatbuffers IDL actually solving the described painpoints.