rerun-io/rerun #2143: Python SDK rehaul to fit blueprints

nikolausWest, 2023-05-16:
The introduction of SDK control of the Blueprint means the introduction of a new major concept to keep track of for the user of the SDK. It now becomes even harder to keep track of the difference between data and style/configuration in the mind of the user.

With blueprint, we now have two databases for users to put things:

- The Entity Hierarchy database (ends up in the rrd)
- The View Hierarchy (blueprint)

There are some constraints but you can basically put both data and styling/config in both of these places. We also currently mix data and styling freely in the Python SDK, which makes it harder to keep track of what is happening for the user, and also makes the SDK less composable.

Before adding lots more graphical primitives we should clean up the api's to make things more clear to both us and the users.

emilk, 2023-05-26:
We have a plan for this involving objects for datatypes, components and archetypes. It needs writing down though.

Wumpf, 2024-02-26:
@emilk didn't we have a writeup for this somewhere on notion or is this issue now just hopelessly outdated? :)

abey79, 2025-03-05:
Closing as "hopelessly outdated".

closed 2025-03-05 by abey79