rerun-io/rerun #2167: CI:

emilk, 2023-05-17:
We have a nice `PR Build Summary` for each of our PRs, but still a lot of web-related bugs slip through.

I think a good next step is to include a link to `app.rerun.io` (or `demo.rerun.io`) directly in the PR description, with a checkbox, e.g.:

* [ ] I have tested https://app.rerun.io/commit/be4014c/index.html

This is one less click, and a reminder to us to test our PRs

emilk, 2023-06-01:
Actually, it would probably be better to keep `demo.rerun.io` user-facing, and instead publish to e.g. `test.rerun.io` or similar

jprochazk, 2023-06-29:
Should we create a new issue for part 2? (`test.rerun.io`)

jprochazk, 2023-06-29:
I created a separate issue, because `test.rerun.io` is a separate task (splitting `demo.rerun.io` into "fast examples to test with" and "something to show off")

closed 2023-06-29 by jprochazk