rerun-io/rerun #2178: Error setting Ctrl-C handler: MultipleHandlers

albx79, 2023-05-21:
**Describe the bug**
Running the cargo-installed CLI tools panics immediately, with the error in the title.

**To Reproduce**
1. `cargo install rerun`
2. `RUST_LOG=debug rerun`
3. see error: ```

thread 'main' panicked at 'Error setting Ctrl-C handler: MultipleHandlers', rerun-0.5.1/src/run.rs:635

   6: core::panicking::panic_fmt
             at core/src/panicking.rs:64:14
   7: core::result::unwrap_failed
             at core/src/result.rs:1750:5
   8: rerun::run::setup_ctrl_c_handler
   9: rerun::run::run::{{closure}}
  10: tokio::runtime::park::CachedParkThread::block_on
  11: tokio::runtime::scheduler::multi_thread::MultiThread::block_on
  12: tokio::runtime::runtime::Runtime::block_on
  13: rerun::main

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
[2023-05-21T17:07:30Z DEBUG re_analytics::sink_native] {
      "api_key": "phc_sgKidIE4WYYFSJHd8LEYY1UZqASpnfQKeMqlJfSXwqg",
      "batch": [
        {
          "timestamp": "2023-05-21T17:07:30.093360341Z",
          "event": "crash-panic",
          "distinct_id": "d49da5f1-f9b5-4bbb-90b5-3e83ea1e1972",
          "properties": {
            "git_hash": "v0.5.1",
            "debug": false,
            "llvm_version": "15.0.7",
            "target": "x86_64-unknown-linux-gnu",
            "event_id": 1,
            "file_line": "rerun-0.5.1/src/run.rs:635",
            "session_id": "19ba9d47-294a-4abd-95cf-0ef410d94c2e",
            "build_date": "2023-05-21T12:49:02Z",
            "rust_version": "1.69.0 (84c898d65 2023-04-16)",
            "callstack": "   6: core::panicking::panic_fmt\n             at core/src/panicking.rs:64:14\n   7: core::result::unwrap_failed\n             at core/src/result.rs:1750:5\n   8: rerun::run::setup_ctrl_c_handler\n   9: rerun::run::run::{{closure}}\n  10: tokio::runtime::park::CachedParkThread::block_on\n  11: tokio::runtime::scheduler::multi_thread::MultiThread::block_on\n  12: tokio::runtime::runtime::Runtime::block_on\n  13: rerun::main",
            "rerun_version": "0.5.1",
            "rerun_workspace": false
          }
        }
      ]
    }

```
**Expected behavior**
Either show a window, or a user-friendly error message to explain what I did wrong.

**Screenshots**
N/A

**Backtrace**
<!-- If this is a crash, include the backtrace in a code-block (```[STACKTRACE]```). -->
```[STACKTRACE]
   6: core::panicking::panic_fmt
             at core/src/panicking.rs:64:14
   7: core::result::unwrap_failed
             at core/src/result.rs:1750:5
   8: rerun::run::setup_ctrl_c_handler
   9: rerun::run::run::{{closure}}
  10: tokio::runtime::park::CachedParkThread::block_on
  11: tokio::runtime::scheduler::multi_thread::MultiThread::block_on
  12: tokio::runtime::runtime::Runtime::block_on
  13: rerun::main

```

**Desktop (please complete the following information):**
Linux Fedora 38
`Linux localhost.localdomain 6.2.14-300.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Mon May  1 00:55:28 UTC 2023 x86_64 GNU/Linux`

**Rerun version**
```
rerun 0.5.1 [rustc 1.69.0 (84c898d65 2023-04-16), LLVM 15.0.7] x86_64-unknown-linux-gnu, built 2023-05-21T12:49:02Z
```

**Additional context**
Using the runner as a library from my rust program works fine. This problem happens only with the runner installed by cargo.

emilk, 2023-05-22:
Thanks for the report! This works on `main`, so this will be fixed by the upcoming 0.6.0 release

emilk, 2023-05-25:
This could still happen for users using `rerun` as a dependency, but it should be properly fixed by https://github.com/rerun-io/rerun/pull/2198 now

closed 2023-05-25 by emilk