rerun-io/rerun #2179: Logging (partially) infinite primitives

roym899, 2023-05-22:
**Is your feature request related to a problem? Please describe.**
Sometimes it could be useful to log a 3D plane or lines that extend to infinity. A closely related feature would be primitives that extend to infinity only in some directions (e.g., rays). 
An example would be to visualize the epipolar plane of two images.

**Describe the solution you'd like**
Either separate functions such as `log.plane` / `log.ray` / `log.line` or adding support for `np.inf` (or similar) to existing functions.

**Describe alternatives you've considered**
Simply logging a very long line behaves reasonably well already, but isn't as nice. Logging a plane seems only possible indirectly through `log_rect` + `log_rigid3` right now. 

**Additional context**
Some loosely related issues: #1361, #1531

emilk, 2023-05-22:
Logging planes and rays seems reasonable. Infinite lines could be supported with [Plücker Coordinates](https://geometricalgebra.org/plucker_coordinates.html), though I have a hard time comping up with an example for that

roym899, 2023-05-22:
I agree, there don't seem to be too many use cases for lines, but, I guess, a grid or axes could be constructed with those.