rerun-io/rerun #2180: Add close / hide buttons to view toolbars

roym899, 2023-05-22:
**Describe the annoyance**
To hide a currently visible view, I need to either remove or hide it from the left-hand side blueprint view. 
Especially when there are many views or the names are not unique, I often have to click on the name in the toolbar first, to highlight the view in the the blueprint view. Adding an extra click and some extra mouse movement.

**Expected behavior**
It would be easier if at least hide (and possibly remove) were also visible in the toolbar of the view.

**Screenshots**
[image]

emilk, 2023-05-22:
I agree, this would be nice

roym899, 2023-05-22:
After seeing your reply on #2181 I wonder if those buttons should actually be per-tab (similar to browser tabs). Otherwise it's unclear whether closing a window would close all tabs or not.

Wumpf, 2024-02-15:
@abey79 I figure this might become obsolete with the context menus + '...' context menu button that I think was planned?