rerun-io/rerun #2181: Clicking anywhere in a view's titlebar should select it

roym899, 2023-05-22:
**Describe the annoyance**
Currently to select a view you have to click on the label of the view. I believe it might be more intuitive if the whole titlebar would select the view, similar to how selecting windows in operating systems work. I often click on the titlebar, just to notice it doesn't do anything.

More debatable, but maybe also clicking anywhere inside the view (that is, not on one of the primitives) should select the respective view. However, I'm not sure if that would cause to many undesired selection changes.

roym899, 2023-05-22:
Oh, I didn't notice this on my screen. I agree that makes it less clear.

nikolausWest, 2023-05-30:
Perhaps we should extend the tab to the whole titlebar when there is only one tab + make the tabs more visually distinct?

nikolausWest, 2023-05-30:
From Mårten:

_1. I agree that a single tab should span the whole space.
2. The Figma designs differ in 2 ways from the implementation:
    1. They use title case for the tabs
    2. They use icons for the tabs
These 2 things make the tabs feel like a thing and not just like a random text label. I think we should implement these things before doing any other fixes._

I agree with this as a start so let's go with that

Wumpf, 2024-02-15:
I think we sufficiently improved the tabs and space view selecting behavior to close this (:

closed 2024-02-15 by Wumpf