rerun-io/rerun #2186: Selected timeline heuristic is unstable

teh-cmc, 2023-05-23:
Try it yourself by running `examples/python/text_logging` a bunch of times: sometimes `log_time` wins, sometimes `frame_nr` wins, depending on the speed of ingestion.

jleibs, 2023-07-27:
I've observed this leading to confusion on multiple occasions -- bumping priority to high so that it actually gets fixed.

Also marking as bug as this is a departure from our expected behavior of always defaulting to user-provided timeline.

The major challenge with implementing this is that there is a race-condition when we start laying out views where it's possible that the only data we have received is the `SetStoreInfo` message and thus have no timelines to base our decision off of.  Similar to other properties we probably need some form of an `EditableAutoValue` here.

nikolausWest, 2023-08-07:
Solution: 
- make it an `EditableAutoValue`
- auto = first user timeline

PR #2986 (2023-08-16): Automatically select user timeline if no timeline was explicitly selected yet
* Fixes #2186

Slight deviation from the desired behavior from the ticket: We'll always select the user timeline which is _alphabetically first_. I deem this good enough though and an improvement in any case!

Going back the the "auto select" state is not possible explicitely, but will happen if for whatever reason no user timeline is available anymore.

Tested as seen in the video and then confirmed separately the described alphabetic behavior

https://github.com/rerun-io/rerun/assets/1220815/0d3303d7-7b19-4854-993d [... cut]
commit d0de66c5 (2023-08-16): Automatically select user timeline if no timeline was explicitly selected yet (#2986)

* Fixes #2186

Slight deviation from the desired behavior from the ticket: We'll always
select the user timeline which is _alphabetically first_. I deem this
good enough though and an improvement in any case!

Going back the the "auto select" state is not possible explicitely, but
will happen if for whatever reason no user timeline is available
anymore.

Tested as seen in the video and then confirmed separately the described
alphabetic behavior

https://github.com/rerun-io/rerun/assets/1220815/0d3303d7-7b19- [... cut]
commit 751936ed (2023-08-17): Automatically select user timeline if no timeline was explicitly selected yet (#2986)

* Fixes #2186

Slight deviation from the desired behavior from the ticket: We'll always
select the user timeline which is _alphabetically first_. I deem this
good enough though and an improvement in any case!

Going back the the "auto select" state is not possible explicitely, but
will happen if for whatever reason no user timeline is available
anymore.

Tested as seen in the video and then confirmed separately the described
alphabetic behavior

https://github.com/rerun-io/rerun/assets/1220815/0d3303d7-7b19- [... cut]
commit c852e2d4 (2023-08-31): Automatically select user timeline if no timeline was explicitly selected yet (#2986)

* Fixes #2186

Slight deviation from the desired behavior from the ticket: We'll always
select the user timeline which is _alphabetically first_. I deem this
good enough though and an improvement in any case!

Going back the the "auto select" state is not possible explicitely, but
will happen if for whatever reason no user timeline is available
anymore.

Tested as seen in the video and then confirmed separately the described
alphabetic behavior

https://github.com/rerun-io/rerun/assets/1220815/0d3303d7-7b19- [... cut]
closed 2023-08-16 by Wumpf