rerun-io/rerun #2200: Make the repository less unix-centric

jprochazk, 2023-05-24:
A lot of tooling in this repository assumes that:
- `bash` exists (and that the user's shell understands hashbangs)
- symlinks just work ™️

I think removing these two assumptions will make the repository much more accessible to Windows users.
- [ ] `scripts/setup*` don't work on Windows, meaning they can't easily install required tooling
- [ ] `re_web_viewer_server` is configured to fetch the `web_viewer` files via a symlink which doesn't work when the repository is freshly cloned
- [ ] a lot of `just` recipes use bash (such as `py-build`, which is central to everything regarding the `rerun` wheel)

More investigation needs to be done to find all the other things which don't work on Windows out of the box.

emilk, 2023-06-07:
The instructions for Windows were improved in https://github.com/rerun-io/rerun/pull/2319

jprochazk, 2023-11-28:
@Wumpf what's your experience like now? Are there any remaining snags?

I've had great success using pixi on Windows/Mac for everything. It Just Works™️, and I haven't had to manually run the contents of any bash script yet.

Wumpf, 2023-11-28:
Yeah let's close this. There's still some details that don't work like for example roundtrip test scripts aren't function yet on Windows and various scripts under /scripts/ don't work yet. `just` works by now but isn't great (looking forward to remove it completely)
The symlink in re_web_viewer isn't reaaaally a problem if git is configured correctly on windows. There's a note [on BUILD.md](https://github.com/rerun-io/rerun/blob/main/BUILD.md#building-for-the-web) by now

closed 2023-11-28 by Wumpf