rerun-io/rerun #2217: Weird looking arg-parser logic in minimal example

jleibs, 2023-05-25:
The minimal example now contains the code:
```
_, unknown = __import__("argparse").ArgumentParser().parse_known_args()
[__import__("logging").warning(f"unknown arg: {arg}") for arg in unknown]
```
with no comment or other explanation.

I believe this may be a workaround to keep things from failing in conjunction with the run-all script, but this seems like the wrong approach. This complexity should be moved into the runner, not obscuring the example code.

nikolausWest, 2023-05-30:
We should really try to avoid making our examples less clear in order to make it easer for devs to run all examples. Better to put that complexity in our helper scripts

emilk, 2023-05-30:
100%. We should move this into the `run_all.py` script

closed 2023-06-13 by emilk