rerun-io/rerun #2222: API not indexed by search engines

roym899, 2023-05-25:
**Describe the annoyance**
Right now it seems nearly impossible to find direct links to the Python API with google / bing / etc.

**To Reproduce**
Some search terms that would not come up with anything:
`rerun log_image`, `rerun log_image python api`, `rerun python api`, ...

Typically the docs.rerun.io pages show up, however, direct links to ref.rerun.io seem quite difficult to find.

jprochazk, 2023-06-14:
Seems like #2404 helped, as Google now reports that the live URL is indexable. We're just waiting on it to actually crawl and index the pages now.
[image]

emilk, 2023-07-07:
googling "site:ref.rerun.io" is now giveing nine hits! So some progress :)

Wumpf, 2023-07-18:
I'm only seeing 8 now 🤔

Wumpf, 2025-01-30:
Looks like we're fairly well indexed by now