rerun-io/rerun #2229: All files Rerun can open should support command line, drag, and open from menu

nikolausWest, 2023-05-26:
Since 0.6, Rerun supports opening png images, meshes and rrd files form the command line, but only rrd files can be opened through the Rerun menu -> open flow or by dragging and dropping into the viewport. All files that Rerun can open should be accessible through the same methods.

emilk, 2023-08-17:
This affects the text on the welcome screen:

[image]

abey79, 2023-08-17:
I'm getting that when dropping files on demo.rerun.io:

[image]

abey79, 2023-08-17:
Also, `preview_files_being_dropped()` appears to be inoperative on wasm builds.

abey79, 2023-08-18:
Currently, CLI loading of files uses SmartChannels, where as Open/file drop use `SystemCommand::LoadRrd`. The latter is bad, because it hogs the SystemCommand thread and doesn't "stream" the RRD into a recording–it loads all of it at once. This could be bad for big files.

It appears that the way forward is:
- reimplement `SystemCommand::LoadRrd` by creating a smart channel (`App` will need some kind of `Arc<HashSet<Receiver<_>>>`)
- add new system command for other types of files (jpg, etc.), reusing the existing support in `loading.rs`
- it'll probably be cleaner to move the `load_rrd_file_to_channel` from `rerun/src/run.rs` to `re_viewer/src/loading.rs`?
- re-implement the file arguments to CLI by using the new `SystemCommand` mechanism, which will allow to still have a Tcp server when launching a viewer with file arguments (basically it'll decouple the two features).

Thanks to @jleibs for helping me gain clarity on this.

abey79, 2023-08-21:
This is going in the direction of #2121

emilk, 2023-08-25:
Me, @abey79 and @jleibs discussed this at length with the following conclusions:

We need something like this:

``` rust
enum DataSource {
    /// A remote RRD file, served over http.
    RrdHttpUrl(String),

    /// A path to a local file.
    #[cfg(not(target_arch = "wasm32"))]
    FilePath(std::path::PathBuf),

    /// Used mostly on Web where there are no paths
    FileContents(FileContents),

    /// A remote Rerun server.
    WebSocketAddr(String),
}
```

which replaces `ArgumentCategory` in `run.rs`. There will be a `SystemCommand` to load it.

We then have a function like so:

```rs
fn stream_data_source(ds: DataSource) -> Result<smart_channel::Receiver<LogMsg>> {
    …
}
```

that will be called by `run.rs` but also by `app.rs.`

The other change is that `App` stores a `RecieverSet` which is something like `struct RecieverSet(Vec<Receiver<LogMsg>>)` which handles pruning closed channels.

teh-cmc, 2023-08-26:
The issue title mentions CLI, drag-and-drop and open-from-menu; but what about URLs?

E.g. it's possible to do `https://app.rerun.io/?url=https://somewebsitething.com/data.rrd` today, and it should be equally possible to do `https://app.rerun.io/?url=https://somewebsitething.com/image.jpeg`.

abey79, 2023-08-26:
Opening from URL is indeed also required in the short-term for the welcome screen's example launcher (#3096).

closed 2023-08-29 by emilk