rerun-io/rerun #2238: `RecordingStream::set_time_*` are not automatically used as documented

kpreid, 2023-05-28:
**Description**
The Rust documentation says that the time in `RecordingStream::{set_time_sequence, set_time_seconds, set_time_nanos}` will be “used for all subsequent logging performed from this same thread”. However, it is not. The timeline name I pass to `set_time_sequence()` does not appear in the viewer.

**Rerun version**
`0.6.0`

**Additional context**
Based on examining the code, it appears that the only effect they have is to change the output of `RecordingStream::now()`, but `now()` is not called automatically by anything outside of the `python_bridge` code. Perhaps it should be automatically included in a similar fashion as the `log_time` timeline?

I have found that calling `.with_timepoint(stream.now())` on each `MsgSender` causes the set time to appear in the viewer, which suffices as a workaround.

emilk, 2023-06-27:
Thanks for the report!

This should be fixed now by:
* https://github.com/rerun-io/rerun/pull/2506

closed 2023-06-27 by emilk

Wumpf, 2023-06-27:
Fixed in #[2506](https://github.com/rerun-io/rerun/pull/2506)