rerun-io/rerun #2240: Share X axis between multiple scalar plots

asuper0, 2023-05-29:
**Is your feature request related to a problem? Please describe.**

When deal with datas of long time, we may need to zoom in for every plot, which is troublesome to operate and could make the axis of different plots not aligned.

**Describe the solution you'd like**

I suggest to add a feature to the viewer which work like matplotlib ` plt.subplots(3,1, sharex='col') `. The argument `sharex = 'col'` can share the x-axis between subplots, and always keep the same while zooming, dragging the plots.

emilk, 2023-06-27:
This is very similar to:
* https://github.com/rerun-io/rerun/issues/294

abey79, 2025-04-23:
Concrete proposal for this: https://github.com/rerun-io/rerun/issues/9789