rerun-io/rerun #2254: Make sure all developers are on Python 3.8

emilk, 2023-05-29:
Python 3.8 is our minimum supported version. Everything that works in 3.8 is likely to work in future Python version (Python is good at backwards compatibility).

The problem is that sometimes we developers add new features for e.g. 3.11 and then only notice once it fails on CI in cryptic ways, or worse, fails at run-time for things that aren't covered by CI.

I'm not sure how we enforce a specific Python version though.

abey79, 2023-05-29:
FYI, [MacPorts](https://www.macports.org) will let you install multiple (major) versions of Python interpreters without fuss. It's been my go-to Python installation method for ages. Here is what I currently have installed:

```
$ python3.8 --version
Python 3.8.16
$ python3.9 --version
Python 3.9.16
$ python3.10 --version
Python 3.10.11
$ python3.11 --version
Python 3.11.3
```

emilk, 2023-08-08:
homebrew also supports this:

```
❯ brew search python@3.
==> Formulae
boost-python3    python@3.10 ✔    python@3.11 ✔    python@3.7       python@3.8       python@3.9 ✔

❯ python3.10 --version
Python 3.10.12
```

adeebshihadeh, 2024-09-02:
We use `uv` for openpilot and love it. It replaced pyenv, pip, and poetry for us. 

https://github.com/astral-sh/uv

abey79, 2024-10-21:
Closing this because CI has been invented so that developper aren't force to use outdated tools :)

closed 2024-10-21 by abey79