rerun-io/rerun #2266: Upgrade example documentation

nikolausWest, 2023-05-30:
Many people learn best from studying examples, but our documentation around our examples is currently sub-par. We need to make it much easier to find the most helpful example and make the documentation easier to maintain.

- [ ] #2259 
- [ ] Reorganize example docs structure
- [ ] Add section to examples docs with external examples
- [ ] Unify example documentation with the examples README(s)
- [ ] Categorize examples by tags
- [ ] Unify example screenshots
- [ ] All example screenshots should have alt-texts
- [ ] Docs site shows what rerun api's are used in each example
- [ ] Examples docs shows live web viewer where possible

emilk, 2023-05-30:
It's high priority to work on these issues, but not high priority to close every single on of them. Let's start with the low hanging fruit

nikolausWest, 2023-08-07:
Closing this issue and moving the remaining pieces we want to do to to a new tracking issue

closed 2023-08-07 by nikolausWest