rerun-io/rerun #2273: Unify example screenshots

nikolausWest, 2023-05-30:


emilk, 2023-05-30:
Related:
* https://github.com/rerun-io/rerun/issues/2117

emilk, 2023-06-12:
Figure out workflow and document it:
* Use the new screenshot feature: https://github.com/rerun-io/rerun/pull/2293
* Standardized sizes
* pngcrush everything?
* CSS for a dropshadow so the images don't merge with our HTML background

jprochazk, 2023-06-12:
We should have `scripts/upload_image.py` produce a few different screenshot widths (while preserving the aspect ratio):
- `480px`
- `768px`
- `1024px`
- `1200px`
- original

Instead of outputting a single link, the upload image script will output a responsive image element snippet using the generated :

```
<picture>
  <!-- responsive: -->
  <source media="(max-width: 480px)" srcset="https://static.rerun.io/HASH_FILENAME_480w.png" />
  <source media="(max-width: 768px)" srcset="https://static.rerun.io/HASH_FILENAME_768w.png" />
  <source media="(max-width: 1024px)" srcset="https://static.rerun.io/HASH_FILENAME_1024w.png" />
  <source media="(max-width: 1200px)" srcset="https://static.rerun.io/HASH_FILENAME_1200w.png" />
  <!-- fallback: -->
  <img src="https://static.rerun.io/HASH_FILENAME_orig.png" />
</picture>
```

This can still be directly pasted into a markdown file, and it should work in all browsers, becase unrecognized tags are treated as `div` tags.

abey79, 2023-06-13:
With automatic image stack creation and support for reading from clipboard in #2411, the workflow now looks like:

1) run the example
2) resize the viewer to some reasonable size (that is legible but smaller than fullscreen on your 4K/5K screen)
3) groom the blueprint until it looks nice
4) take a screenshot via the command palette
5) `just upload --name my_example_name`
6) Copy/paste the resulting HTML in the example's README

I think this workflow gets us pretty far until more automation with blueprints is possible. We _could_ consider formalising (2) a little more, but I'm not sure the technical pain of enforcing this is worth the value.

nikolausWest, 2023-06-13:
Sounds great!

abey79, 2023-06-16:
Closed by #2440

closed 2023-06-16 by abey79