rerun-io/rerun #2274: Image viewer normalizes (?) the colors and popup shows unnormalized RGB

roym899, 2023-05-30:
**Describe the bug**
```python
import rerun as rr
import numpy as np
import matplotlib.pyplot as plt

rr.init("image color", spawn=True)
img = np.random.random((300, 300, 3)) * 0.2
rr.log_image("image", img)
print(img.min(), img.max())  # to check they are in range [0,1]

plt.imshow(img)
plt.show()
```
gives
[image]

Hovering over the image in rerun shows low RGB values, but bright colors.

**Expected behavior**
I'd expect it to look the same / very similar with the shown RGB values being the same as those in the raw data.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
```
rerun_py 0.7.0-alpha.0+ [rustc 1.69.0 (84c898d65 2023-04-16), LLVM 15.0.7] x86_64-unknown-linux-gnu main b263b9f, built 2023-05-29T18:38:22Z
```
Also happens in `0.6`

nikolausWest, 2023-05-31:
Ran into this one as well. Kind of sucks

jleibs, 2023-05-31:
Noticed this when reporting: https://github.com/rerun-io/rerun/issues/2295

Noticed matplotlib and rerun didn't show the same colors:
[image]

emilk, 2023-06-01:
When a user logs images with float colors, we have to make some assumption about the color range.

The current behavior is to find the min and max value and normalize to that. `matplotlib` seems to assume a 0-1 color range. This may seems reasonable (and is probably a good default) but it opens a lot of questions:

* What about values outside the 0-1 range - should we clamp those? Or show them in a special "error" color?
* What if the user logs values in the 0.0-255.0 range - if we assume a 0-1 range things will look very bright
* Sometimes users want to log wild ranges, like dept maps, and then normalizing makes perfect sense

We should expose all this to the user in the blueprint, and we should have saner defaults. I suggest:

* We add a `DataRange` component for tensors, with a min-max range
* When `DataRange` is missing, we infer it using the following heuristic:
  * If all finite values are in 0-1 range, assume 0-1 range
  * If all finite values are in 0-255 range, assume 0-255 range
  * Else, set the `DataRange` to the actual min/max values

----

Then there is the problem of gamma (I think this might be what you are running in to @jleibs). This should also be made into a component so that users can specify either `Linear` or `sRGB Gamma` (and eventually more options), but for now the same assumption is that all data is in gamma space, and we should make that assumption.

roym899, 2023-06-01:
I think for `log_image` it's okay to either expect 0-255 / 0-65535 / 0.0-1.0 depending on the data type of the passed array (similar to what the docs suggest right now). I think clamping + printing a warning seems like a reasonable behavior for floats outside this range.

For depth maps there is `log_depth_image`. For other data that should be colormapped based on min and max there is `log_tensor`.

I agree though adding control over the min and max values would be nice especially for `log_tensor`.

nikolausWest, 2023-06-02:
The matplotlib heuristic (and therefore what most ML / vision people are likely to expect) is:
- If integer valued (int, uint, 8, 16, 32, 64) expect image data in 0-255 range
- If float valued expect image data in 0-1.0 range
[image]

teh-cmc, 2023-06-02:
I very much like @emilk's proposal: it's elegant, intuitive and it scales. And AFAICT it would even meet matplotlib's users' expectations.

We could also apply the same logic for 16bit+ integer tensors, e.g. for a `u32` tensor whose `DataRange` component is missing:
1. try 0-255
2. try 0-65535
3. try min-max

PR #2342 (2023-06-12): Automatically determine image/tensor color mapping & need for srgb decoding

### What

Fixes #2274 
* #2274 

Will need to be further worked on via:
* https://github.com/rerun-io/rerun/issues/2341

Introduces 3 new things actually:
* finite range, determined if the range of a tensor was in fact not finite
* heuristic for color mapping range
* heuristic for srgb
   * ⚠️ does this break things?

Range & sRGB:

This
```
import rerun as rr
import numpy as np
import matplotlib.pyplot as plt

rr.init("image color", spawn=True)
img = np.random.random((300, 300, 3)) * 0.2
rr.log_image("image", img)
pr [... cut]
commit 15cb22c6 (2023-06-12): Automatically determine image/tensor color mapping & need for srgb decoding (#2342)

### What

Fixes #2274 
* #2274 

Will need to be further worked on via:
* https://github.com/rerun-io/rerun/issues/2341

Introduces 3 new things actually:
* finite range, determined if the range of a tensor was in fact not
finite
* heuristic for color mapping range
* heuristic for srgb
   * ⚠️ does this break things?

Range & sRGB:

This
```
import rerun as rr
import numpy as np
import matplotlib.pyplot as plt

rr.init("image color", spawn=True)
img = np.random.random((300, 300, 3)) * 0.2
rr.log_image("image", [... cut]