rerun-io/rerun #2285: Slowdown with 1'000 entities

emilk, 2023-05-31:
When we have 1k entities the viewer starts slowing down due to a score of places where we do store lookups for each entity each frame.

Some of these store lookups is due to scene ingestion: every entity is checked against every archetype ("is this a point? is this a mesh? is this an arrow?"). This `O(N*K)` complexity needs to be solved.

Furthermore, these queries are taking up way too much time, they could be _a lot_ faster.

We also need a good example/test of this to work from.

teh-cmc, 2023-05-31:
> Some of these store lookups is due to scene ingestion: every entity is checked against every archetype ("is this a point? is this a mesh? is this an arrow?"). This `O(N*K)` complexity needs to be solved.
>
> Furthermore, these queries are taking up way too much time, they could be a lot faster.

I believe this is the result of us addressing the problem backwards at the moment.
I.e. we should be asking the store "hey, i have this archetype: which entities might possibly be concerned at all?", rather than that `O(nk)` query we're doing right now.

In fact the store might actually already be capable of answering that question efficiently, iirc.

emilk, 2023-08-21:
@jleibs brought up the idea of entity tagging, which is very relevant for this. Each archetype logged to an entity adds a tag for that archetype (e.g. its name). This means each entity has a set of archetype tags (usually only one tag).

emilk, 2023-08-21:
A good reproduce is running `examples/python/open_photogrammetry_format/main.py --no-frames` and then hiding the points (the `world/pcl` entity).

[image]

Enhance:
[image]

emilk, 2023-08-21:
Me, @Wumpf, @jleibs and @teh-cmc discussed this, with a bunch of conclusions.

The first low-hanging fruit we should do is just to speed up the mapping, without changing any behavior or ui.

For each space view we pre-calculate a lookup map of `ComponentName` -> `ViewPartSystemId` like so:

```rs
let mut component_view_parts : HashMap<ComponentName, Set<ViewPartSystemId>> = {}
for view_part in all_view_part_categories {
    for required in view_part.required_components() {
        component_view_parts[required].insert(view_part.id());
    }
}
```

This will produce a lookup looking something like:

```json
{
    "rerun.mesh": {"mesh"},
    "rerun.point3d": {"point3d", "arrow"},
    "rerun.transform": {"transform_arrows"},
     …
}
```

Note that the cardinality of the `Set<ViewPartSystemId>` is usually very low, and we may want to optimize based on this.

Using this we then each frame categorize the entities of a Space View like so:

```rs
let mut view_part_entities: Map<ViewPartSystemId, Vec<Entity>> = {}

for entity in space_view.entities {
    let entity_comps = db.all_components(entity);

    let mut potential_parts: Set<ViewPartSystemId> = {}
    for component_name in entity_comps {
        potential_parts = potential_parts.union(potential_parts[component_name]);
    }

    for part_id in potential_parts {
        let view_part_system = all_view_part_systems[id];
        if view_part_system.required_components().all(|comp| db.entity_has_component(comp)) {
            // It is a hit!
            view_part_entities[part_id].push(entity);
        }
    }
}
```

And then we have what we wanted: for each view part system, we have a list of entities for that.
Will this be faster? Probably. It can also be trivially parallelized (e.g. using `rayon`).

-----

This same idea can later be reused to speed up the broader query of "What `SpaceViewClass`es does this Entity fit?" which we use as a heuristic for generating Space Views, and populating them with any new entity.

emilk, 2023-08-21:
In the future we want to be able to expose to users a setting for "What ViewParts should be used to show this Entity?". For instance, given a mesh with a transform, should we show it as a mesh, a transform, or both? This would require a `BTreeMap<EntityPath, EditableAutoValue<HashSet<ViewPartSystemId>>>` stored in each `SpaceView`, and a UI for that. This would also replace the special checkbox for wether or not to show the transform-axis for an entity.

Wumpf, 2023-08-22:
Got this quite a bit faster, however we figured out in the process that as long as auto-adding space views is enabled that takes a significant amount of time. @jleibs and me discussed how to improve the way we do space view heuristics in the future and came up with a new schema where each space view class would decide on its own heuristic (this has the added benefit of removing remaining knowledge of specific views from `re_viewport`. The heuristic method per class would look something like this:

```rust
    fn auto_spawn_heuristic(
        &self,
        ctx: &ViewerContext<'_>,
        ent_paths: &PerSystemEntities,
        max_num_candidates: usize,
    ) -> Vec<(AutoSpawnHeuristic, EntityPath)>;
```

meaning that candidate creation can be done in a much more sensible way. Some challenges there are dealing with disconnected-components.

Wumpf, 2023-08-22:
separated this out into https://github.com/rerun-io/rerun/issues/3079

emilk, 2023-08-25:
`cargo r -p raw_mesh` is also a pretty good benchmark

closed 2023-08-29 by jleibs