rerun-io/rerun #2289: Better default window size when running in notebooks

nikolausWest, 2023-05-31:
When displaying Rerun in a notebook, the default window size is way too big. We should use a smaller size by default. Pyplot uses a default figsize default: 6.4 x 4.8 inches so I suggest we go with that for starters. 

Related (perhaps should be a different issue):
Would be great if the window could be sized to fit the data (when that makes sense) so that we don't show unnecessary "dead" areas
[image]

nikolausWest, 2023-06-01:
> If we show selection/blueprint we need all the space we can get.

Thus: https://github.com/rerun-io/rerun/issues/2288

Wumpf, 2024-07-09:
@jprochazk @jleibs now that you come back fresh from notebook land: Is it realistic for us to do this in the future or are the controls we have for dynamically sizing the widget too limited?

jprochazk, 2024-07-09:
There's no restriction to how we can size the widget. We just need to update `DEFAULT_WIDTH` and `DEFAULT_HEIGHT` in `notebook.py`.