rerun-io/rerun #2295: Always allow image to zoom out to 1:1 scale

jleibs, 2023-05-31:
When displaying an image that is smaller than the current canvas, there is no way to "zoom out" to the native 1:1 resolution of the image.  This means small images may be oversized.

Exa:
[image]

There is no way to scale down the image here to match the size of the imshow window without doing something like abusing the selection panel to take up space.

emilk, 2023-06-12:
This is part of a larger issue of rewriting the 2D "eye" (view) code to allow arbitrary panning and zooming

emilk, 2023-08-08:
Blocked by
* #2531

emilk, 2024-05-06:
It's unclear how this should work. We could add a button to go to `zoom=1` for a 2D space view, but I'm not sure how useful that would be. Should it keep that zoom even if we resize the space view? Would it be one image pixel per screen pixel, or per ui point?

jleibs, 2024-05-06:
Maybe could see what other common image preview tools do for this.