rerun-io/rerun #2305: Api for saving screenshots and animations

chisarie, 2023-06-02:
Hello! 

It would be great if there was a way to log a pointcloud or mesh to rerun, specify a camera pose, and programmatically save a "screenshot" of that object from that specified pose. It would be very useful for evaluating and comparing different things, and for publishing purposes. 

An extension of this would be to provide a list of camera poses, and use it to generate multiple screenshots that can be then stitched together to make a video or gif.

From looking at the docs, it seems this functionality is not there yet, is it?

Thank you in advance for the help!

nikolausWest, 2023-06-02:
Hey @chisarie! Thanks for the question / feature request. Screenshots from the command line actually just landed (#2293). The ability to construct the Rerun Viewer from a declarative spec is not there yet, but under way (see PR with RFC here: https://github.com/rerun-io/rerun/pull/2302). Once that lands with the ability to also set a 3D view's eye I believe most of what you want should be there

chisarie, 2023-06-02:
Sounds great! One small thing to note which is different than https://github.com/rerun-io/rerun/pull/2293 is that I'd like to have a "screenshot" of a viewport alone, without all the gui around it. Looking forward to it :)

chisarie, 2023-06-02:
Maybe screenshot is not the right word for this, what I mean is more like an offscreen renderer, similar to Open3d's and Pyrender's ones: 
http://www.open3d.org/docs/latest/python_api/open3d.visualization.rendering.OffscreenRenderer.html, 
https://pyrender.readthedocs.io/en/latest/examples/offscreen.html

nikolausWest, 2023-06-04:
Sure thing. We have the pieces in place to screenshot single views without UI surroundings as well. I'm not familiar enough with the offscreen render in Open3D to know how that works exactly but if you also need the rendering to run headless that would be a bit further off with Rerun.

Would you mind sharing a bit more background around how and why you would intend to use a feature like that?

chisarie, 2023-06-07:
Mainly to generate comparison pictures to use on publications paper and website. For example to compare the outputs of different methods. If I have to take all screenshots manually it would be very time consuming and also not precise (camera position should always be the same to compare properly). To use rerun for this, another important feature would also be to be able to set a white background, see https://github.com/rerun-io/rerun/issues/754

nikolausWest, 2023-06-07:
Thanks @chisarie, that makes a lot of sense!