rerun-io/rerun #2315: Distortion model for allowing fisheye lens entity projection

pablovela5620, 2023-06-05:
**Is your feature request related to a problem? Please describe.**

Currently only a pinhole perspective camera is supported, in order to project points from 3d views into 2d for views that have significant distortion such as fisheye lens one needs to do it manually for accurate projection
**Describe the solution you'd like**

Addition of distortion parameters as described [here](https://xrprimer.readthedocs.io/en/latest/data_structure/camera.html#fisheye) (same as what OpenCV uses)
**Describe alternatives you've considered**

Current alternative involve just doing the projection yourself

**Additional context**

[image]
Here you can see how the current implementation leads to inaccurate projections

nikolausWest, 2023-06-07:
Also realized we're missing an issue for adding support for regular Brown-Conrady lens distortion

pablovela5620, 2023-06-12:
Wanted to add that this would also probably require arctan projection (I haven't worked with fisheye cameras before, realized that arctan projection is an important component to making the projection correct)

nikolausWest, 2023-06-21:
Here is a good tutorial on computer vision with fisheye models for the eventual implementer: https://plaut.github.io/fisheye_tutorial/#the-fisheye-projection

dgriffiths3, 2024-02-09:
Any update on this? Would love to see this feature.

nikolausWest, 2024-02-09:
Hey @dgriffiths3 - great to hear this feature would be useful for you! Do you think you could expand a bit on exactly which parts you'd hope to see in support for fisheye models?

Regarding update, we've been focused on all the work leading up to launching blueprint (slotted for release after 0.13). After that we'll shift back much more to adding more datatypes and models etc. Currently brown-conrady distortion models are probably slightly higher up in priority but it's not impossible we do both together.

dgriffiths3, 2024-02-09:
Hi @nikolausWest, that's great to hear it's on the horizon! For my current use case we plot 3D boxes in world space, and then visualise them from the egocentric camera view. We use a combination of both approx. pinhole and high distortion (fisheye) lenses. Similar to the ARKitScenes example you have, but with addition cameras. 

It would be expected that a straight box edge would be quite curved when viewed on the fisheye image, but currently this is not the case.

I think it might even be more useful to initially just provide an API for users to easily add their own custom camera model class. I think no matter how many models you support, there will always be another.

jlazarow, 2024-05-04:
@nikolausWest Would it make sense to add some sort of archetype (or indicator?) for Spatial2DView's that caused an additional Fisheye distortion (i.e., a warping shader on the image itself) on top of the existing Pinhole rendering results? My understanding is that a general Fisheye renderer is not always want people actually expect (i.e., it will probably work well on high-resolution meshes/point clouds but be less than desirable on say, 3D boxes). For some other wgpu-based work (wgpu-py), I already did this and it is generally is pretty satisfactory.

Wumpf, 2024-05-04:
My current thinking is that we do it as a distortion as you say. Meaning it's essentially a postprocessing pass. The big disadvantage with that is that we may need to render a higher resolution to begin with otherwise we're getting sampling artifacts. Also, it's unclear if we can do it for the camera visualization in 3D easily: I believe we'll probably limit it to be active when "taking the viewport of the camera" since this is much more straight forward.
Doing it as a postprocessing step should make it invariant on whether we're dealing with high resolution meshes or less tessellated things like boxes, so I'm not sure that is what you meant? 🤔 

As to where to specify it: depends all on the actual data modelling. It's a property of a camera so I'd expect it to live with the camera, but I'm less sure if that's a new archetype or if it should be crammed into the Pinhole one as an optional "extension" so to speak.

jlazarow, 2024-05-04:
Right, I think we're pretty aligned (post-processing). I agree on the downside being having to render at a higher resolution (although perhaps this factor would be a function of the degree of distortion). From a (personal) priority aspect, I was more indicating that:

1. Assuming the 3D points are coming from an unprojection of a fisheye sensor at some instant of time, perhaps it's less interesting to expect to freely navigate the point cloud through the view of a fisheye sensor (but maybe this is important for self-driving applications?).
2. Rather, the ability to view the world structures (e.g., 3D boxes) rendered against a set of actual camera poses collected from a device (e.g. the ARKitScenes example) and overlayed on the image itself (which presumably is already captured with distortion) might narrow the computational burden heavily and would be independent of what's actually being rendered.

ShenghaoZhou, 2024-12-04:
Adding to this issue, is it possible to allow user to specify the mapping between pixels and ray direction? This will allow users to use whatever camera model they want. The mapping stays the same for the same camera, so it only needs to be computed once.

pablovela5620, 2025-02-27:
wanted to add this as it may be helpful
https://facebookresearch.github.io/projectaria_tools/docs/tech_insights/camera_intrinsic_models

astraw, 2025-03-31:
FWIW, there is already some rust code for generating calibrations at https://github.com/powei-lin/camera-intrinsic-calibration-rs and I also wrote a little crate to support the Extended Unified Camera Model (EUCM) at https://github.com/strawlab/extended-unified-camera-model.

Wumpf, 2025-08-05:
@gmedan nice, thanks for sharing!
copied your notebook over to [here](https://colab.research.google.com/drive/1U-bgCTHXbLiqiiqf9_jg9GB064LvtrYy?authuser=1#scrollTo=fh1vpN9j0SDd) in order to make the pip install work more out of the box (I had some issues with it)

pablovela5620, 2025-12-06:
> Also realized we're missing an issue for adding support for regular Brown-Conrady lens distortion

One more thing to add here, would be great if we could also include Kannala Brandt, not sure where this issue is right now on the priority list, but I have two different cameras I'm working with where distortion parameters matter alot (on using Brown-Conrady and the other Kannala Brandt) . I could just log the undistorted image, but this isn't ideal since I'm usually working with video, I would also love to include the actual distortion parameters when saving an RRD file (which I guess right now I could use AnyValues or DynamicArchetype to create a custom pinhole camera).

maxwell-cavnue, 2026-09-22:
@paulygreen we tried the undistortion work on [mtl-ai/rerun@pg/distortion_plus_render](https://github.com/mtl-ai/rerun/tree/pg/distortion_plus_render) and it works well for us:

- Cherry-picked 49a62d6 and 19cb46c onto 0.38.1 with no content conflicts
- Tested the Web viewer build, WebGL2, real 4K camera video (AssetVideo + VideoFrameReference)
- Matches our cv2.initUndistortRectifyMap  / ffmpeg preprocessing pipeline that we are constrained to atm.
- One gotcha: null coefficients make the whole Pinhole chunk fail to decode silently. Zero-filling fixes it.

Do you plan to upstream this? Happy to help test.

alcolado, 2026-09-23:
@EricPedley I agree with your 2 cents! I made a similar comment in this other thread about distortion https://github.com/rerun-io/rerun/issues/2499#issuecomment-4972444907. I think the branch mentioned above might address your use case (full disclosure, I work with the author of that branch).