rerun-io/rerun #2335: app.rerun.io crashes on mobile safari

emilk, 2023-06-08:
On some iOS devices, app.rerun.io crashes on mobile safari after about one second. The crash shows no error message, and the site just re-loads.

Perhaps out-of-memory?

teh-cmc, 2023-06-08:
Same exact behavior for me on firefox iOS.

emilk, 2023-06-09:
I did catch this:

```
[Error] RuntimeError: Unreachable code should not be executed (evaluating 'wasm._dyn_core__ops__function__FnMut_____Output___R_as_wasm_bindgen__closure__WasmClosure___describe__invoke__h7151addea9e051c0(retptr, arg0, arg1)')
	<?>.wasm-function[std::alloc::rust_oom::hc4a15e24f24fdc8f] (re_viewer.js:238:137)
	<?>.wasm-function[__rust_alloc_error_handler]
	<?>.wasm-function[alloc::alloc::handle_alloc_error::rt_error::h2c73e9f1de42a5a3]
	<?>.wasm-function[alloc::alloc::handle_alloc_error::hd73a554e7696031d]
	<?>.wasm-function[jpeg_decoder::worker::compute_image_parallel::h139c8b75d60c14d3]
	<?>.wasm-function[jpeg_decoder::decoder::compute_image::h77938edab6ca94eb]
	<?>.wasm-function[jpeg_decoder::decoder::Decoder<R>::decode_planes::hdeda1a493788ebce]
	<?>.wasm-function[jpeg_decoder::worker::WorkerScope::get_or_init_worker::he31a9589035074b3]
	<?>.wasm-function[jpeg_decoder::decoder::Decoder<R>::decode_internal::h104e2639f2857781]
	<?>.wasm-function[<image::codecs::jpeg::decoder::JpegDecoder<R> as image::image::ImageDecoder>::read_image::h030481f710262c69]
	<?>.wasm-function[image::image::decoder_to_vec::h8e1655fba1af6194]
	<?>.wasm-function[image::dynimage::decoder_to_image::h5847b645c5ec126c]
	<?>.wasm-function[<image::io::free_functions::load_inner::LoadVisitor as image::io::free_functions::DecoderVisitor>::visit_decoder::h73308b4b5ffc8c4e]
	<?>.wasm-function[image::io::free_functions::load_decoder::h350b07cd04f449e4]
	<?>.wasm-function[image::io::reader::Reader<R>::decode::h756876518fcd91d4]
	<?>.wasm-function[re_components::tensor::DecodedTensor::try_decode::h00706fd57bab661e]
	<?>.wasm-function[re_viewer_context::tensor::tensor_decode_cache::TensorDecodeCache::entry::h873b8e309c756124]
	<?>.wasm-function[re_viewer_context::caches::Caches::entry::h411e805b1999ae62]
	<?>.wasm-function[re_space_view_spatial::scene::parts::images::ImagesPart::process_entity_view::h3123ee9248b13e15]
	<?>.wasm-function[re_space_view_spatial::scene::parts::entity_iterator::process_entity_views::h784333263cdcc0e4]
	<?>.wasm-function[<re_space_view_spatial::scene::parts::images::ImagesPart as re_viewer_context::space_view::scene_part::ScenePart<re_space_view_spatial::space_view_class::SpatialSpaceView>>::populate::hd7abdf77881e0bd1]
	<?>.wasm-function[<core::iter::adapters::flatten::FlatMap<I,U,F> as core::iter::traits::iterator::Iterator>::next::h1470a1dba4173704]
	<?>.wasm-function[<alloc::vec::Vec<T> as alloc::vec::spec_from_iter_nested::SpecFromIterNested<T,I>>::from_iter::h2afdb50a16ad5e8c]
	<?>.wasm-function[<re_viewer_context::space_view::scene::TypedScene<C> as re_viewer_context::space_view::scene::Scene>::populate::h88940fc5523f047d]
	<?>.wasm-function[re_viewport::space_view::SpaceViewBlueprint::scene_ui::h129b7de333f935c9]
	<?>.wasm-function[<re_viewport::viewport::TabViewer as egui_tiles::behavior::Behavior<re_viewer_context::SpaceViewId>>::pane_ui::h664b9f9100f8d559]
	<?>.wasm-function[egui_tiles::tree::Tree<Pane>::tile_ui::hf14bc9e637e86f2b]
	<?>.wasm-function[egui_tiles::tree::Tree<Pane>::tile_ui::hf14bc9e637e86f2b]
	<?>.wasm-function[egui_tiles::container::linear::Linear::vertical_ui::h495feff51cb34d85]
	<?>.wasm-function[egui_tiles::tree::Tree<Pane>::tile_ui::hf14bc9e637e86f2b]
	<?>.wasm-function[egui_tiles::tree::Tree<Pane>::ui::hbd13c1fd3a693298]
	<?>.wasm-function[core::ops::function::FnOnce::call_once{{vtable.shim}}::h775070d4e1189a15]
	<?>.wasm-function[<alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once::h8a9940c27ef7ca3d]
	<?>.wasm-function[egui::ui::Ui::scope_dyn::h8723a432c95dcd0e]
	<?>.wasm-function[re_viewport::viewport::Viewport::viewport_ui::h789ced57005a8a9e]
	<?>.wasm-function[core::ops::function::FnOnce::call_once{{vtable.shim}}::h69396589fc443429]
	<?>.wasm-function[<alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once::h8a9940c27ef7ca3d]
	<?>.wasm-function[core::ops::function::FnOnce::call_once{{vtable.shim}}::h1521b4cc4e9cfbf0]
	<?>.wasm-function[egui::containers::frame::Frame::show_dyn::h5a18230eddc59c8d]
	<?>.wasm-function[egui::containers::panel::CentralPanel::show_inside_dyn::ha3aad74b145f506b]
	<?>.wasm-function[re_viewer::ui::blueprint::Blueprint::blueprint_panel_and_viewport::hae50f70fb939728d]
	<?>.wasm-function[core::ops::function::FnOnce::call_once{{vtable.shim}}::hf9dc021e50a7ff28]
	<?>.wasm-function[<alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once::h8a9940c27ef7ca3d]
	<?>.wasm-function[core::ops::function::FnOnce::call_once{{vtable.shim}}::h1521b4cc4e9cfbf0]
	<?>.wasm-function[egui::containers::frame::Frame::show_dyn::h5a18230eddc59c8d]
	<?>.wasm-function[egui::containers::panel::CentralPanel::show_inside_dyn::ha3aad74b145f506b]
	<?>.wasm-function[re_viewer::app_state::AppState::show::h4eb9142a056c6a25]
	<?>.wasm-function[core::ops::function::FnOnce::call_once{{vtable.shim}}::h675813dcd33cdcc6]
	<?>.wasm-function[<alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once::h8a9940c27ef7ca3d]
	<?>.wasm-function[core::ops::function::FnOnce::call_once{{vtable.shim}}::h1521b4cc4e9cfbf0]
	<?>.wasm-function[egui::containers::frame::Frame::show_dyn::h5a18230eddc59c8d]
	<?>.wasm-function[egui::containers::panel::CentralPanel::show_inside_dyn::ha3aad74b145f506b]
	<?>.wasm-function[egui::containers::panel::CentralPanel::show_dyn::h52d2ef37069c5843]
	<?>.wasm-function[<re_viewer::app::App as eframe::epi::App>::update::h880c1502cbacd1c3]
[... cut ...]
	<?>.wasm-function[eframe::web::app_runner::AppRunner::logic::h9c7c3f7076ccf7c5]
	<?>.wasm-function[<T as wasm_bindgen::closure::WasmClosureFnOnce<(),R>>::into_fn_mut::{{closure}}::h6cd07725ec6d4f9f]
	<?>.wasm-function[<dyn core::ops::function::FnMut<()>+Output = R as wasm_bindgen::closure::WasmClosure>::describe::invoke::h7151addea9e051c0]
	wasm-stub
	__wbg_adapter_47 (re_viewer.js:238:137)
	real (re_viewer.js:209)
```

This is very surprising, because the fiat demo only uses 450 MB of VRAM plus ~200 MB of RAM, according to our own bookeeping.

emilk, 2023-06-14:
It's usually fine if I pause it just as it loads, and then continue it later.

Wumpf, 2024-02-13:
@emilk is this bug still around?

emilk, 2024-02-13:
yes