rerun-io/rerun #2353: Expose connection status in the Python API

abey79, 2023-06-09:
The Python API should expose the connection status and/or a disconnect event when streaming and the remote viewer is closed. For live streaming applications (e.g. based on video capture, etc.), this would enable the script to automatically stop streaming when the viewer is closed by the user.

Wumpf, 2025-04-29:
The underlying `LogSink` traits should have ways to report their status and optionally (?) have hook for waiting status change.

For example something that is very confusing today is that `connect_grpc` for instance is not blocking until a connection has been established and will rather continuously try to establish one. Neither is there an indication when it is done, nor is there a way to block on it nor is there a way to tell when the viewer is disconnected.

fortminors, 2025-06-28:
@Wumpf Are there any plans on implementing this feature?

Wumpf, 2025-06-29:
@fortminors I tried giving it a "quick" spin, was a bit more involved than I thought it would be.
Would be awesome if you could have a look and check if that's useful for you https://github.com/rerun-io/rerun/pull/10433

fortminors, 2025-06-30:
@Wumpf Thank you! I have checked it and it works - I am able to intercept the connection loss, however rerun itself stopped working... :)
I continuously get errors in rerun, saying 
```
ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column
```
It seems that it's a different issue because the rest of the code is left unchanged. I was using rerun `0.23.2` before and now I am trying a fresh one on your branch

fortminors, 2025-06-30:
@Wumpf I double checked with my previous code that does not use your feature, everything works fine with `0.23.2` and `0.23.4`. It fails with the same error when I build rerun from source, as in adding that line to my cargo.toml
```
[dependencies]
rerun = { git = "https://github.com/rerun-io/rerun.git" }
```
I will open a new issue for that since it is unrelated.

fortminors, 2025-06-30:
The issue is solved using latest rerun viewer. Are there any plans in merging this into main branch? Or are you searching for other options right now to expose the status? 
Thank you!