rerun-io/rerun #2385: Document improvements (meta)

abey79, 2023-06-12:
Quick and dirty issue to list "early user" documentation issues from my experience.

Coverage holes:
- `rr.log_cleared` missing from Python ref landing page: #2348 
- Annotation context / class description / annotation info entirely missing from the concepts section
- Although `rr.log_annotation_context` is linked from the Python Common API page, there is no link/description for `rr.ClassDescription` and `rr. AnnotationInfo`.
- Blueprint stuff entirely missing from the concepts section
- Blueprint stuff entirely missing from the Python common API page (although an "Experimental" section already exists)

Structure:
- all API mentioned in docs should link into refs

More subjective stuff TBD:
- Wording around annotation/class stuff "sounds" way more generic that it actually is.
- Use "book" style to introduce concepts
  - each chapter builds on previous chapter, rather than introducing concepts in isolation and seemingly random order
  - each chapter contains more code examples
  - Next/prev buttons, etc. to encourage a more systematic reading (as opposed to the current "trust your instincts, read specific topic when needed" approach.

emilk, 2023-08-08:
Let's clean up what's left to do here

abey79, 2023-08-18:
I think we can close this as everything is either:
- done
- properly logged as issues scheduled for planning consideration
- obvious/very long-term/likely to be revisited anyway stuff