rerun-io/rerun #2403: Annotation context display is unstable with multiple classes

abey79, 2023-06-13:
**Describe the annoyance**

When annotation contexts have multiple classes, their display is flickering due to random ordering of the classes.

https://github.com/rerun-io/rerun/assets/49431240/2a3f025e-701b-43a9-b202-808116ba1a6d

**To Reproduce**
Steps to reproduce the behavior:

This demo RRD: 
[demo.rrd.zip](https://github.com/rerun-io/rerun/files/11732017/demo.rrd.zip)

Also, the `mp_face` example (#2338) can be used:

```python
python examples/python/mp_face/main.py --demo-image
```

In the timeline, click "video/landmarker/annotation_context" to display the glitch.

emilk, 2023-07-04:
This is really weird. The UI comes from `ClassDescription::keypoint_connections` which is defined as:

```rs
pub keypoint_connections: Vec<(KeypointId, KeypointId)>
```

So there is no hashmap there. It must somehow get populated in a random order, but I don't immediately see how.

abey79, 2023-10-17:
I can't reproduce anymore, this has likely been fixed.

closed 2023-10-17 by abey79