rerun-io/rerun #2439: Freeze pip package versions

emilk, 2023-06-15:
We have a bunch of `requirements.txt` files which specify packages without version constraints, e.g:

```
numpy
pillow
requests>=2.31,<3
rerun-sdk
timm==0.6.11
torch>=1.13.0
transformers
```

We should probably make sure that every dependency is either specified with `==` or with `>=X,Y`.

A lint could check all `requirements*.txt` files and make sure each line contains either `==`, or `>=` and `<`

abey79, 2023-06-15:
This might create other issues, for example breaking on newer versions of python because the pinned version is wasn't then compatible. I believe the right way to go about this is:
- use a package management tool that allows for lock files (poetry, hatch, etc.—whichever plays the nicest with maturin)
- be more systematic with testing examples on all supported python version (with e.g. tox or nox)

abey79, 2023-06-15:
Side note: a tool like poetry would be really helpful to manage multiple dev environment to easily switch from 3.8 to 3.11, etc.

emilk, 2024-01-11:
Relevant for the upcoming numpy 2.0 release: https://pythonspeed.com/articles/numpy-2/

emilk, 2024-04-10:
Closed as duplicate of
* https://github.com/rerun-io/rerun/issues/2253

closed 2024-04-10 by emilk