rerun-io/rerun #2445: CI: systematic testing of SDK and examples on all supported (major) Python version

abey79, 2023-06-15:
We currently support all current Python version from 3.8 to 3.11, and will probably aim to support 3.12 ASAP when it's out. Backward compatibility is pretty good on the language side but dependency management can be tricky. For example, pinning versions might fix an issue in 3.8 but break compatibility with, e.g. 3.11 because that version of the package wasn't yet compatible with 3.11.

Regardless of how we want to address that in the future, systematic testing of both the SDK and examples on all supported Python version is a prerequisite.

This issue is about:
- figuring out the right tool for systematic testing (tox, nox, GH action matrix, ...)
- deploying said tool on the SDK and examples vs. Python 3.8 to 3.11

emilk, 2024-03-01:
We should run `nox -s tests` on nightly at least

abey79, 2024-10-21:
Duplicated by #7510 (which has now more content, some I'm closing this one)

closed 2024-10-21 by abey79