rerun-io/rerun #2478: Add helper class for pinhole's intrinsics matrix

abey79, 2023-06-19:
**Describe the annoyance**

Creating the pinhole intrinsics matrix with a raw numpy array is annoying, especially in tutorials and examples (e.g. [here](https://github.com/rerun-io/rerun/blob/4a5a4c6ad6ab86763481c2c6b40257f78d29a798/docs/code-examples/depth_image_simple.py#L18). We should provide an helper class that would create the matrix from focal length/width/height, or field-of-view, etc.

nikolausWest, 2023-06-21:
I think we should rather add more Pinhole camera representations, like we do with Transform3D, so that we can log the pinhole representation we have, and see the same data when we inspect the entity in the viewer

abey79, 2024-10-21:
Done in https://github.com/rerun-io/rerun/pull/3372

closed 2024-10-21 by abey79