rerun-io/rerun #2494: 2D point radius != None shows up inconsistently in 2D and 3D view

roym899, 2023-06-20:
**Describe the bug**
Logging 2D points to an image plane they show up as huge 3D points (pixels converted to meters?). Things look fine as long as `radii=None`.

**To Reproduce**
```python
import numpy as np
import rerun as rr

rr.init("projection", spawn=True)

rr.log_view_coordinates("world/camera", xyz="RDF", timeless=True)

rr.log_pinhole(
    "world/camera/image",
    child_from_parent=np.array([[500, 0, 250], [0, 500, 250], [0, 0, 1]]),
    width=500,
    height=500,
)
rr.log_image("world/camera/image/rgb", np.ones((500, 500, 3)))
rr.log_points("world/camera/image/points", np.random.uniform(0, 500, (30, 2)), radii=1)
```

**Expected behavior**
[image]

**Actual behavior**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
```
rerun_py 0.7.0 [rustc 1.69.0 (84c898d65 2023-04-16), LLVM 15.0.7] x86_64-unknown-linux-gnu prepare-0.7 9cf3033, built 2023-06-16T15:47:32Z
```

roym899, 2023-06-20:
Maybe related to #2279.

emilk, 2023-06-26:
It seems all points in 2D views have too small radius, at least on retina displays

abey79, 2023-11-09:
This issue has been encountered and reported by some of our users. IMHO, since the 2D point radius correspond to "pixel" units in the 2D space view, and these pixels have a defined size in the 3D space view (on the pinhole widget projection plan), we should be able to adjust the 3D radius to be consistant. That would involve taking into account both the pinhole project itself and the "Image plan distance" parameter.

#### Expected

[image]

#### Actual

[image]

abey79, 2023-11-09:
Fun fact: setting the "Image plan distance" to the same value as the pinhole focal yields the correct behaviour:

[image]

SeaOtocinclus, 2023-11-09:
@abey79  FYI, it seems to happen too if you are logging a Boxes2D, so the problem is most likely happening for more than one archetype.

abey79, 2023-11-10:
Indeed, and the glitches are... interesting:

[image]

Again, like for points, better handling of "Image plan distance" might improve on this issue:

[image]

<details>
<summary>Code</summary>

```python
import numpy as np
import rerun as rr

rr.init("rerun_example_pinhole")
rr.connect()

rng = np.random.default_rng(12345)

image = rng.uniform(10, 30, size=[200, 300, 3])
rr.log("world/image", rr.Pinhole(focal_length=300, width=300, height=200))
rr.log("world/image", rr.Image(image))
rr.log("world/image/point", rr.Points2D([50, 70], radii=5))
rr.log("world/image/box", rr.Boxes2D(centers=[100, 100], half_sizes=[20, 30], radii=2))
```
</details>

PR #4199 (2023-11-13): Fix line & points (& depth clouds points) radii being unaffected by scale & projection via Pinhole
### What
Longstanding issue!

* Fixes #1223
* Fixes #1219
* Fixes #2494
* Replaces #4196

Shader only fix, in the future the scale factor shouldn't be extracted on the fly for every vertex out of the transform and instead passed in, but I wanted to keep the change minimal. The added vertex shading cost is unlikely to matter all that much _short term_.
(also was very nice iterating on this and get before/after screenshots ;))

Throw-away test script for this:
```py
import numpy as np
import rerun  [... cut]
commit 6ee3c6af (2023-11-13): Fix line & points (& depth clouds points) radii being unaffected by scale & projection via Pinhole (#4199)

### What
Longstanding issue!

* Fixes #1223
* Fixes #1219
* Fixes #2494
* Replaces #4196

Shader only fix, in the future the scale factor shouldn't be extracted
on the fly for every vertex out of the transform and instead passed in,
but I wanted to keep the change minimal. The added vertex shading cost
is unlikely to matter all that much _short term_.
(also was very nice iterating on this and get before/after screenshots
;))

Throw-away test script for this:
```py
import numpy as np
impo [... cut]
closed 2023-11-13 by Wumpf