rerun-io/rerun #2497: Add orthographic and axis lock features to the 3D view

abey79, 2023-06-21:
An orthographic projection of a 3D scene with well-defined axes can sometimes usefully complement our current user-controlled perspective view. For example, this would enable replicating the perspective/front/left/top views of CAD software: 

[image]

This would be extending the 3D view with options to:
- set the projection to ortho
- partially restrict the view-eye control, e.g. locking it's orientation (the user would retain the ability to pan and zoom)
- fully restrict the view-eye, for a total control of the display

Note that a 3D view with ortho + locked axes is technically similar to our 2D viewer, so [extending it](https://github.com/rerun-io/rerun/issues/683#issuecomment-1600377999) instead might enable similar end results.

Potentially relates to:
- #683 
- #1510
- #349

SeaOtocinclus, 2023-12-08:
Would be very useful for all mapping related display!

mont1004, 2024-07-10:
Hi there!

I'm really looking forward to this feature. While having projection with well-defined axes is nice, just being able to switch to ortho projection is highly valuable for me. If it's relatively simple to implement just that feature, would it be possible to introduce just the projection method switching functionality a bit earlier?

mwopfner, 2025-04-17:
This would be a great help to better interpret some data (like @SeaOtocinclus mentioned).
My use case is to get a top down view of some lidar data and a map so that I can see if the lidar points align with the wall points in the map. The perspective projection is fooling me if I'm not looking from straight above.

mwopfner, 2025-04-17:
It looks like the `Eye` struct already have the notion of an orthographic camera and only the `ViewEye` lacks an appropriate `Orthographic` `EyeMode` implementation.
If i read the source code correct the `SpatialView3D` uses the information of an `Eye` and creates a `TargetConfiguration` with an `Projection`. In the `SpatialView3D` currently only `Perspective` support is implemented but the `Projection` struct has an Orthographic mode.
So the most things are already in place and "just" must be put together. 😉

Wumpf, 2025-04-21:
Most of the work is to make picking and the wgsl shaders in re_renderer aware of orthogonal perspective. Most things should just work out of the box, but there's various things that aren't aware of orthogonal cameras yet (although I think most of re_renderer is already)