rerun-io/rerun #2514: rr.serve doesn't change port

mix0z, 2023-06-26:
When you use rr.serve with custom ports the browser viewer continues listening to default port

**To Reproduce**
Steps to reproduce the behavior:
1. Use rr.serve with custom web_port and ws_port
2. Try to open in browser 
3. You will get re_viewer.js:1681 WebSocket connection to 'ws://adress:9877/' failed: 

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 for serving and mac OS X for opening in browser
 

**Rerun version**
rerun_py 0.7.0 [rustc 1.69.0 (84c898d65 2023-04-16), LLVM 15.0.7] x86_64-unknown-linux-gnu prepare-0.7 9cf3033, built 2023-06-16T15:47:32Z

emilk, 2023-06-27:
Thanks for the report! This is probably an easy fix - we'll try to get to it soon

Wumpf, 2023-06-27:
@mix0z I have trouble reproducing that issue. I just tried with `rec.serve(True, 5002, 6002)` which correctly spawns a browser with http://localhost:5002?url=ws://localhost:6002

Did you maybe open http://localhost:5002 instead? That would then indeed to back to the default web socket port. Otherwise, can you try refreshing the browser and see if that helps?

mix0z, 2023-06-27:
@Wumpf Yes, sorry, i just opened wrong address

closed 2023-06-27 by mix0z