rerun-io/rerun #2523: Return iterators rather than vectors when deserializing in Rust backend

teh-cmc, 2023-06-26:
The deserialization routines for components and datatypes in the Rust backend currently return full-on `Vec`s, while there is in fact nothing preventing us from returning lazy iterators.

Current:
```rust
pub trait Datatype {
    // [...]

    fn from_arrow_opt(data: &dyn ::arrow2::array::Array) -> Vec<Option<Self>>
    where
        Self: Sized;

    // [...]
}

pub trait Component {
    // [...]

    fn from_arrow_opt(data: &dyn ::arrow2::array::Array) -> Vec<Option<Self>>
    where
        Self: Sized;

    // [...]
}
```

`impl trait` is still unstable when used in a return position in traits, but it wouldn't be too much work to add an associated type, especially now that we have GATs! e.g.:
```rust
pub trait Datatype {
    type Iterator<T> = /* ... */

    // [...]

    fn from_arrow_opt(data: &dyn ::arrow2::array::Array) -> Self::Iterator<Option<Self>>
    where
        Self: Sized;

    // [...]
}

pub trait Component {
    // [...]

    fn from_arrow_opt(data: &dyn ::arrow2::array::Array) -> Self::Iterator<Option<Self>>
    where
        Self: Sized;

    // [...]
}
```

Of course we could also settle for a boxed trait object (`Box<dyn Iterator<Item = Self>>`) instead, but why do a half arsed job when it's all codegen'd anyway :upside_down_face:

teh-cmc, 2023-07-24:
Update: we now know for a fact that `Box<dyn Iterator>` is not viable.

jleibs, 2023-07-24:
This will be a blocker for 0.9 but doesn't need to block landing into main once we've cut 0.8.

teh-cmc, 2023-07-24:
Also, we need to take care of fallibility, e.g. by using something like https://docs.rs/fallible-iterator/latest/fallible_iterator/index.html

emilk, 2023-08-08:
Probably related: `just py-build --release && ./examples/python/open_photogrammetry_format/main.py` is ~100x slower than it should be (10s/frame)

emilk, 2023-08-09:
If this helps as much as we hope, we can then revert the central commit of https://github.com/rerun-io/rerun/pull/2950 (the added aching)

nikolausWest, 2023-08-17:
Let's discuss if we even need this at all. Most of the performance problems have already been fixed.

emilk, 2023-08-25:
I think we should do this instead:
* https://github.com/rerun-io/rerun/issues/3109

closed 2023-08-25 by emilk